JulianKemmerer/PipelineC #2: Why C?

piranna, 2021-04-12:
I know there are more C-to-HDL compilers in the field, but being a new one, maybe Rust could be a good aproach due to its type safety (I think there are someones based on Go due to its coroutines). Only reason I could think about using C would be pointers... Is there any reason to have used C instead of Rust? Honest question, and just curious :-)

JulianKemmerer, 2021-04-12:
Hi there, thanks for reaching out - and an excellent question:

Why C? Because in 2014 as an FPGA engineer I was in a world where, software folks, who know C, would write up some algorithm. And then hand over that algorithm in C to me the FPGA engineer, who also knows C. So it seemed natural that, if trying to bridge the hardware programability gap to pick a language that is already pretty much the intermediate between software and hardware folks already. Its a simple language and everyone knows it. Also I do not have a theoretical CS / programming languages background (FPGA engineer by training) - so again, the language side of things needed to be dead simple.

Why not Rust? (or others): Since the project started with 'what made sense and was easy' there isnt a hard reason why not another language and I am definitely open to other front end languages.
Ill list a few thoughts though: At the time of starting this project in 2014, Rust was not quite as known and in use as it is now. I do like its type safety and general modern-language feel. Initially - its main feature seemed to the borrow checker / that memory management - which isn't needed/applicable for PipelineC (lacking pointers, dynamic memory allocation). Recently I took a look at an abstract syntax tree for a simple Rust program compared to a C version (I go from language AST -> intermediate logic netlist thing) - and it seemed a bit complicated - getting to my earlier point: I'm not against other languages - its just that they are all more complicated than C and are a nontrivial amount of work to get mapped into my intermediate/netlist/graph thing (which essentially needs to map to VHDL, another constraint). And thats something I'd rather start with some help as opposed to all over again by myself. Also why Ive been investigating alternative intermediates like XLS and CIRCT - so I can completely get out of the front end languages and intermediates game - and just do the backend pipelining and FPGA related stuff.

This is also the sort of answer I give to 'why not make MyNewSpecialC language custom for hardware description?' Its alot of work and Im no expert.

So short answer to 'other languages?' is 'yeah why not? I just need some help :)'

JulianKemmerer, 2021-10-03:
PipelineC++ :eyes: 
https://github.com/JulianKemmerer/PipelineC/discussions/32

JulianKemmerer, 2023-01-18:
Yeah to anyone watching this issue or otherwise

The PipelineC Discord is also a great place to also discuss this in a more conversational way https://discord.gg/Aupm3DDrK2

Down for brainstorming :brain: :+1: :nerd_face: