JulianKemmerer/PipelineC #4: Provide alternative flows for OSS integration

ted-xie, 2021-07-27:
In the [Setup](https://github.com/JulianKemmerer/PipelineC/wiki/Running-the-Tool#setup) instructions, the open source synthesis integration instructions call for the user to manually specify the paths for Yosys and NextPNR. This is really sub-optimal, and can make user patches more annoying to review and/or merge.

A more flexible solution would be to allow more options that can leverage industry-standard flows such as environment variables, i.e. `$YOSYS_PATH`, `$NEXTPNR_PATH`, etc.

JulianKemmerer, 2021-07-27:
Sounds good to me - `industry-standard flows such as environment variables` is that to say $YOSYS_PATH, $NEXTPNR_PATH  are common env variables that folks using the OSS are familiar with / will have set?  (for ex. $XILINX_VIVADO seems to be a 'standard' for Vivado - AWS F1 instances have that set)

I was hoping for a 'unzip the OSS suite zip file and change one line/one env var' kinda feel - but I am not a frequent user of the open source flow so whatever folks prefer/expect we can make happen.

@ted-xie 
Are you a user of the GHDL+Yosys+nextpnr flow?

JulianKemmerer, 2021-09-01:
Immediately kind internet folks have noted that just defaulting to PATH visible executables would make sense as a minimum start

JulianKemmerer, 2021-09-03:
Everything but the open source tools now tries to use executables that are in the user path
https://github.com/JulianKemmerer/PipelineC/commit/f4d368ac61d3c0fefdf62e1451ecabb9b5ec03a3

I still need to work out what executables 'define' a user as having installed Yosys w/ GHDL plugin + NEXTPNR

JulianKemmerer, 2021-09-17:
Hoping to adapt information from new friend over at
https://github.com/JulianKemmerer/PipelineC/discussions/12

JulianKemmerer, 2021-09-19:
And close out with work stemming from https://github.com/JulianKemmerer/PipelineC/pull/15

suarezvictor, 2021-09-28:
Paths to tools shouldn't be fixed, I have a problem where ghdl is not on /usr/local/bin but in /opt/ghdl/bin/
A good way to determine ghdl location is not to use absolute paths, or to just put this on commandline
```
`which ghdl`
```

JulianKemmerer, 2021-09-28:
I agree - I was planning on using 'which'
https://github.com/JulianKemmerer/PipelineC/blob/c67ec4dbc53b995148914be82d03fcd5f2fe8b06/src/SYN.py#L15
on the cxxrtl branch - look forward to trying verilator next

JulianKemmerer, 2021-09-30:
Now uses exes in path if oss cad suite is not installed

closed 2021-09-30 by JulianKemmerer