JulianKemmerer/PipelineC #8: Auto remove output directory

JulianKemmerer, 2021-09-01:
Copy pasting the snippet of docs explaining the current situation:

https://github.com/JulianKemmerer/PipelineC/wiki/Running-the-Tool

If the tool is stopped while running, upon next start it attempts to restart where it left off. Good for large designs / long synthesis sweeps. But bad for other yet-to-be-resolved reasons:
Stopping the tool after code parsing completes means the code will not be parsed again - delete your_file.c.parsed from within SYN_OUTPUT_DIRECTORY directories if code is modified after parsing.
Stopping the tool during synthesis creates a failing synthesis run log - delete that log file from it's SYN_OUTPUT_DIRECTORY before trying again.
If all else fails, try removing the SYN_OUTPUT_DIRECTORY before running the tool as a sanity check.
Ex.rm -r ~/pipelinec_syn_output/*; ./src/pipelinec

Want a way to auto remove this directory (or continue using it correctly) as needed

thirtythreeforty, 2021-09-01:
Continuing discussion from Discord:

I think in general I expect a couple things:
- multiple copies of the pipelinec compiler can run in parallel on different designs, and not notice/walk on each other
- the tool doesn't create files in my home directory (unless I ask it to); places I expect build artifacts to go are either
  - the current directory
  - /tmp

The auto-resume feature is neat! Picking up after Ctrl-C is cool. To make it more robust, you might be able to note the last completed step in a file, then have each step preemptively clean any incomplete artifacts that didn't get "checkpointed."

I don't know enough about it to suggest a method to detect something that has failed permanently. Even printing a notice suggesting the user delete the run directory and start over, might help.

thirtythreeforty, 2021-09-01:
To keep different processes from clobbering the same run directory, a common strategy is to drop a `pipelinec.lock` file that contains the PID of the current process using the directory. Briefly:

When you start up, attempt to open the lock file with `O_EXCL | O_CREAT` mode flags - if this succeeds, you know you atomically created the lock, so write your pid and continue. If it exists, read it and see if that process is still alive. If it's dead, it crashed without being able to clean up, so you can delete that file, and attempt to take the lock again.
When you shut down cleanly, delete the lock file.

thirtythreeforty, 2021-09-02:
I like keeping the existing run if possible, just reinvoking the tool to resume is cool. I get your point about being unique in the current directory.

What about basing the default on the basename of the top file? `build-test2` for test2.c. And of course the user can override.

Tracking whether the file has changed (so you know the build dir is stale) could be real "fun." But that's a secondary consideration, you can use timestamps as an easy first pass at that problem.

JulianKemmerer, 2021-09-02:
Definitely like the 'new dir inside current dir' default behavior.

Feel free to give it a go if ya like @thirtythreeforty 

https://github.com/JulianKemmerer/PipelineC/commit/5a1c91a0d31fc60298db18c0b6f101976e4717b4

closed 2021-09-02 by JulianKemmerer