JulianKemmerer/PipelineC #30: How to support HDL generics/parameterized types+functions, etc?

JulianKemmerer, 2021-10-02:
Seed from my todo list:

Basic versions of this can be done with preprocessor but probably want something better

```
Generics? Use VHDL generics somehow?
  Let __vhdl__ describe VHDL funcs that can # pragmad and used somehow?
  Do const func args as vhdl generics - ex. const loop vars into func - how to progate const values?
  Make any modules with constant inputs that inst fully reduceable a renamed func with consts inside?
  Hacky sort of const prop across module boundaries? ew?
  ^Try to optimize/reduce var ref of all const inputs except mux value into constants-inside module version explicit 'LUT' module
```

suarezvictor mentioned used C++ with LLVM to generate C code perhaps :+1:

JulianKemmerer, 2021-10-02:
This could be literally implemented with VHDL generics but it doesn't need to be an exact copy of the functionality

bartokon, 2021-12-07:
Hello, 
I like using parameters in my IP's. 
It is possible to type function like:
fun(ap_bit<8> a, ap_bit<8> b){}
where "8" is a IP core parameter that could be changed, for example in Vivado block design?

JulianKemmerer, 2021-12-07:
Hey there @bartokon - something like you suggest is exactly what I want - but is not quite clear how to get there yet. 

Maybe its faking it by sending C++ code with templates and stuff ex. `ap_bit<8>` style through LLVM -> spit out a bunch of generated C code without template types.

Interesting to mention wanting to change something like a bit width - in a block diagram for example. Pipelining wise, the PipelineC tool produces hardware to meet timing for a specific known/fixed bit width - if that width changes the design may no longer meet timing :-/ 
Would need to run the PipelineC tool again it seems hmmm