JulianKemmerer/PipelineC #41: How should PipelineC package up results for use in other tools?

JulianKemmerer, 2021-11-07:
ex. csv, json or something idk?

bartokon, 2021-11-07:
CSV as most portable and easy to use  imo.

JulianKemmerer, 2021-11-07:
Talking about some kind of csv pipeline description `input AND output` capability 

As well as better summary of latency,fmax from the tool

JulianKemmerer, 2022-10-28:
Should really produce `vhdl` directory that contains all VHDL files for that final run.

Then existing output `.tcl` and `.qip` files pointing into that adjacent vhdl directory should be easy to package up.

Otherwise end up needing to send whole output directory with log files and many VHDL files - gets big...

JulianKemmerer, 2022-10-28:
Vitis could use AXIS specific mapping of ports

```map_clock clk_300p0_0
map_axi_stream io_and_ctrl_in_data_0 io_and_ctrl_in_valid_0 in_ready_return_output_0 clk_300p0_0 input_stream slave
map_axi_stream out_data_return_output_0 out_valid_return_output_0 io_and_ctrl_out_ready_0 clk_300p0_0 output_stream master
```

Some kind of pragma to mark AXIS interfaces?

JulianKemmerer, 2023-01-19:
^ in the coming months should be able to squeeze in time for some kind of `--xo` option to make vivado produce an .xo file with those axi mappings are needed for the ip export

JulianKemmerer, 2023-02-15:
Might need to be `--xo_axis` specific to saying 'make an AXIS mapped IO .xo file'

I think can infer which signals belong to axis by name `valid` `ready` `data` for now....