JulianKemmerer/PipelineC #47: OSS, GDHL, PYRTL

bartokon, 2021-11-20:
Hi, I have a little problem with running PYRTL.
When I source oss cad suite, pipelineC forces me to specify part and I can't go with PYRTL flow. 

But if I run pipelineC without oss, pipelineC runs PYRTL flow but then GHDL is not detected and it fails to run ;o

any ideas? quick fixes?

JulianKemmerer, 2021-11-20:
Hmmm interesting - not sure if I understand you issue about 'sourcing' oss cad suite?

Currently the only way to use pyrtl is to not specify the FPGA part - a little annoying need to comment out `PART` pragma to get the tool to think you want to use pyrtl part-less estimates. 

Maybe we could make up a part specific to pyrtl?
`#pragma PART "pyrtl-asic-130nm"`

JulianKemmerer, 2021-11-20:
To be clear - the tool isnt setup to run both 
`GHDL->Yosys->nextpnr`
AND
`GHDL->Yosys->pyrtl`
It has to be one or the other - current controlled by part pragma as I mentioned

JulianKemmerer, 2021-11-20:
`GHDL is not detected` I think in this case you need to have ghdl installed and in your path?

or more specifically the ghdl plugin for yosys? Maybe you have one and not the other?

bartokon, 2021-11-20:
> Hmmm interesting - not sure if I understand you issue about 'sourcing' oss cad suite?
I type in console - source /home/bartus/oss-cad-suite/environment then run pipelineC

bartokon, 2021-11-20:
Idk if beter is to change #part or change --yoys_json to --estaminate 130nm

JulianKemmerer, 2021-11-20:
Yeah a command line option to say 'ignore the PART in code' and 'estimate w/ pyrtl /asic 130nm' is reasonable

Or even what makes perhaps more sense is a generic `--part` command line override of the in code `PART` pragma. And then maybe like I said, you'd specify in code or command line  `pyrtl-asic-130nm` as the part

JulianKemmerer, 2021-11-20:
Re: using OSS CAD Suite - I have used that environment sourcing feature. If it adds the yosys, ghdl, nextpnr binaries to your path it should work?

I typically edit the path at the top of https://github.com/JulianKemmerer/PipelineC/blob/d08beb0b19b0bf5b3a160abc3eaa8ef92a67a63b/src/OPEN_TOOLS.py#L16

Near that code you can see it looks for these executables in your path, and assumes some relative dirs to those
```
YOSYS_EXE = "yosys"
NEXT_PNR_EXE = "nextpnr-ecp5"
GHDL_EXE = "ghdl"
```

JulianKemmerer, 2021-11-21:
@bartokon let me know if I hit your issue? / can close this out?

JulianKemmerer, 2021-11-28:
Closing for now unless I hear otherwise

closed 2021-11-28 by JulianKemmerer

bartokon, 2021-11-28:
Yes, good call... I think that would work, but now I don't have much time to test it...

bartokon, 2021-12-05:
@JulianKemmerer 
I have changed the path and it crashed 
"./src/pipelinec
[image]

bartokon, 2021-12-05:
After chaning and then sourcing oss suite. It is working like this:
[image]

JulianKemmerer, 2021-12-05:
So the first error Ive seen before and seems to be an issue with PyRTL not knowing about a certain D flip flop model that I assume yosys spat out?

JulianKemmerer, 2021-12-05:
The second image seems like its just not running PyRTL for some reason ...
Oh I think you need to have `--coarse` in the command line ?

JulianKemmerer, 2021-12-05:
No nvm its not `--coarse` thats the issue (yet)

Hmm so you had pyrtl working but then it stopped?

You should see 
```
================== Adding Timing Information from Synthesis Tool ================================
Defaulting to pyrtl based timing estimates...
Using PYRTL synthesizing for part: None
```

If pyrtl python packages are installed

JulianKemmerer, 2021-12-06:
I just pushed something that _might_ help?

@bartokon can you tell me what you want to happen here? 
If you want to use ghdl->yosys->pyrtl  - double check python packages installed pyrtl and pyparsing i think - and dont specify a PART in the code

If you want to use ghdl->yosys->nextpnr , use OSS CAD Suite of have exes installed/visible on path and then DO specify a lattice PART pragma per examples

JulianKemmerer, 2021-12-07:
To be clear `ghdl->yosys->pyrtl ` does also either require `use OSS CAD Suite or have exes installed/visible on path` 

Essentially the switch tor pyrtl from `ghdl->yosys->nextpnr` mode is to have pyrtl installed and dont supply PART