JulianKemmerer/PipelineC #50: Allow operator overloading

JulianKemmerer, 2021-11-29:
Since operator overloading is easily supported for compile as regular C++ it seems like a reasonable idea to make possible for user types now in plain-C PipelineC. Thanks for the push @suarezvictor!

Maybe?
```c
typedef struct ....my_struct_t;
#pragma OVERLOAD * my_struct_t my_mul_op
my_struct_t my_mul_op(my_struct_t lhs, my_struct_t rhs)
{
...
}
```

Or does this feature wait for PipelineC++ rewritten with clang tools etc?

JulianKemmerer, 2021-11-29:
Instead of pragma specifying user func name - as Victor suggested - what if user was forced to match a compiler-known naming scheme

Ex. name exist like a '+' between a uint8 and uint8 yielding a uint9 is named like
`uint9_t BIN_OP_PLUS_uint9_t_uint8_t_uint8_t(uint8_t lhs, uint8_t rhs)` internally for pipelinec already

JulianKemmerer, 2021-11-30:
^ Ex. @suarezvictor
```
typedef struct complex
{
  float re;
  float im;
}complex;
complex BIN_OP_PLUS_complex_complex_complex(complex l, complex r)
{
  complex rv;
  rv.re = l.re + r.re;
  rv.im = l.im + r.im;
  return rv;
}
complex overload(complex x, complex y)
{
  return x + y;
}
```

suarezvictor, 2021-11-30:
It seems good but consider the return type not part of the naming scheme
since it depends only on the input types and the function prototype already
specify that. Functions cannot be overloaded just by the return type

JulianKemmerer, 2021-11-30:
Yeah that definitely makes sense - the return type doesn't need to be part of the name
Unfortunately it is at the moment
I have to check if it is for any dumb internal required reasons or just arbitrary choice I made...

JulianKemmerer, 2021-11-30:
Caught a dumb case where there was a func named `CAST` (for casting types operator)
And it expect names like `CAST_float_uint32_t` for `(float)a_u32`  - based of the formula `func full name = name + output type + input types`

But I just needed to realize that `name` needs to include the output type - not have it be part of the 'formula' 
and it all makes sense again now

JulianKemmerer, 2021-11-30:
This is working in testing - commit coming hopefully soon
```c
typedef struct complex
{
  float re;
  float im;
}complex;
complex BIN_OP_PLUS_complex_complex(complex left, complex right)
{
  complex rv;
  rv.re = left.re + right.re;
  rv.im = left.im + right.im;
  return rv;
}
complex BIN_OP_PLUS_complex_float(complex left, float right)
{
  complex rv;
  rv.re = left.re + right;
  rv.im = left.im + right;
  return rv;
}
#pragma MAIN_MHZ overload0 1.0
complex overload0(complex x, complex y)
{
  return x + y;
}
#pragma MAIN_MHZ overload1 1.0
complex overload1(complex x, float f)
{
  return x + f;
}
```
Only dumb thing thats here to stay for now is needing to call the args for these funcs `left` and `right` - some dumb assumptions baked into generated vhdl and lots of other stuff all over it seems

JulianKemmerer, 2021-11-30:
Should be working now, pull repo
https://github.com/JulianKemmerer/PipelineC/wiki/Automatically-Generated-Functionality#overload

closed 2021-12-01 by JulianKemmerer