JulianKemmerer/PipelineC #51: Implement Various Math Optimizations

JulianKemmerer, 2021-12-19:
For example:

- Check FP sign bit for < 0.0 compare
- FP * 0.25 is FP shift >> 2 (div by 4) exponent math, and similar 1/pow2 consts
- Integer > 255 check using only msbs, and similar for > 'all lsb bits ones' ~pow2 - 1 numbers
- Make * -1.0 into '-' negate operator for easy FP sign flip (mult by 1.0 should optimize away)

Im sure @suarezvictor will continue with more :nerd_face:

suarezvictor, 2021-12-20:
In regards to integers, all such optimization can be done without issues, in case of floating point operations I suggest to enable all this with a -ffast-math option, that in C compilers allows among other things to break IEEE compliance, ignore math with infinites/NaN or "negative zero", and relax rounding behavior

JulianKemmerer, 2022-03-05:
Per https://github.com/JulianKemmerer/PipelineC/issues/57
Do something better than signed division using absolute value, unsigned div, and sign fix

JulianKemmerer, 2023-09-18:
More math fun :smirk: 
https://www.ed-thelen.org/comp-hist/CRAY-1-HardRefMan/CRAY-1-HRM.html#p3-18

suarezvictor, 2023-12-21:
Implement comparison operators using adders:
a > b
a - b > 0
a + (-b) > 0
a + (~b + 1) > 0
a + ~b + 1 > 0
a + ~b > -1
a + ~b >= 0
!sign(a + ~b)

so a > b is the same as !sign(a + ~b)

then
For the < operator, you can swap operands (b > a)
For a <= b, you can use !(a>b)
And for a>=b, replace by !(b>a)

Thus all comparisons will use just an adder