JulianKemmerer/PipelineC #57: integer division

suarezvictor, 2022-03-05:
It would be good that the tools accepts a/b operation even if b is not a float

JulianKemmerer, 2022-03-05:
Since when does this not work?
 Fp div itself contains and integer divider to work?

Will double check in the morning....

JulianKemmerer, 2022-03-05:
First thing to double check is that the arguments are actually integers , check for math producing floats ex 1/3 likely converts to fp 0.333... if used in compile time calculation (not integer 0) - would need to add casts to keep integer types

JulianKemmerer, 2022-03-05:
To be clear this is an issue with specifically signed integers. Unsigned integer division is supported. Just need to 'turn on' signed integer support too...

JulianKemmerer, 2022-03-05:
Turned on signed division using already working unsigned division
https://github.com/JulianKemmerer/PipelineC/commit/3a357cbffea4b65848a276966e76cc013c97ef79

closed 2022-03-05 by JulianKemmerer