JulianKemmerer/PipelineC #64: Make tool optimize for fewer pipelining registers

JulianKemmerer, 2022-03-18:
Right now the tool inserts registers to meet timing - with no regard to how many registers are being inserted.

Instead: manually find the smallest bitwidth wire ~in the area... always choose to slice fewer wires over more if possible with the same timing characteristics/delays ...

JulianKemmerer, 2022-11-02:
Who knows maybe https://github.com/JulianKemmerer/PipelineC/blob/master/src/DEVICE_MODELS.py will get to this some day?

JulianKemmerer, 2022-12-02:
Closing https://github.com/JulianKemmerer/PipelineC/issues/65 definitely helps reduce register use if you are combining global variables and deep pipelines

suarezvictor, 2022-12-03:
Does it lower resource usage? Any measutement?

JulianKemmerer, 2022-12-03:
Yeah recent commit for #65 looks to save ~50 percent of flip flops on ECP5 ray tracer builds - we should do another attempt making it work

On Artix I measured up to like 87K flip-flops saved depending on how your count shift registers.

See info in the pipelinec discord 👍

suarezvictor, 2022-12-03:
Excellent work

bartokon, 2022-12-03:
What about keccak resource usage? I'm curious 😏

JulianKemmerer, 2022-12-04:
@bartokon for keccak - the particulars of #65 I wouldnt expect to apply much IIRC about keccak 
1) since it doesnt maintain a global state variable wired into the pipeline at various places or 2) doesnt have lots of 'unresolved compile time math'/placing doing math on constants that is mistaken for stuff needing LUTs, etc

There also could be some room to tune some of weird pipelining magic numbers ... I am experimenting with some things right now for the ray tracer

So ~maybe but probably not much improvement

JulianKemmerer, 2023-10-29:
Similar issue is wanting to optimize for area

starts with first parsing area out of the reports from the tool...

Similar to #45