JulianKemmerer/PipelineC #74: Make path delay runs arbitrarily parallelizable

JulianKemmerer, 2022-04-07:
The first half of tool execution is getting path delays for modules.

This should be able to occur in any order / parallel.

Currently however, the tool traverses the module hierarchy up the tree synthesizing bigger and bigger modules as it goes.

This behaves very poorly if there are lots of modules near the top level of the design (ex. if top level has a few-sub wrapper levels, you are essentially sequentially synthesizing the same module over and over...)

Whos got time to wait around - not @suarezvictor :-p

suarezvictor, 2022-04-07:
Considering that the aim of the tool is "faster development", it seems a bit important

JulianKemmerer, 2023-01-06:
Also important to note that the second half of tool runtime, iterating on pipelines could also be parallelized...

^good next step after this easier issue is resolved

JulianKemmerer, 2024-10-06:
Fix in https://github.com/JulianKemmerer/PipelineC/commit/ba7ffc2da4549ba7a8b750a73b05365ec64d2b6a

closed 2024-10-06 by JulianKemmerer

suarezvictor, 2024-10-07:
Hi Julian, were you able to measure some gain factors with any specific example?