JulianKemmerer/PipelineC #87: Easier bit manipulation syntax

JulianKemmerer, 2022-07-10:
As opposed to fancier C++ style syntax for bit manipulation, ex `bits<5,2>(x)` as suggested from https://github.com/JulianKemmerer/PipelineC/discussions/32
(trying to replace [current bit manip syntax](https://github.com/JulianKemmerer/PipelineC/wiki/Automatically-Generated-Functionality#bitmanip)...)

Do something that is immediately parseable by `pycparser` and also overloadable by full C++ software compilers (as evidenced by other C++ HLS tools using syntax like below):
```c
uint32_t x;
...
uint1_t b = x[3]; // Fourth bit of x(31 downto 0)
uint16_t y = x(15, 0); // (15 downto 0) 16b slice of x
```

Thanks @Quarky93 !

JulianKemmerer, 2022-07-11:
The bit slice range ex. `(15,0)` must evaluate to a compile time constant range for now... Otherwise this gets into random access/arrays as in https://github.com/JulianKemmerer/PipelineC/issues/89 and is a good deal harder to implement in full...

For now only allow constant index, read only, function syntax (not array `[ ]` based) bit slicing like so:
```c
uint32_t x; // x(31 downto 0)
uint1_t y = x(15); // y = x[15]
uint16_t z = x(15, 0); // z = x[15:0]
```

JulianKemmerer, 2022-07-11:
@Quarky93 are you able to work with that as stated above? Only the function call syntax + constant indices for bit slicing?

JulianKemmerer, 2022-07-12:
OK the new bit slicing syntax is live.
https://github.com/JulianKemmerer/PipelineC/commit/c78752dca5b6df074fe849062cf0c981a15fd116

@suarezvictor you might like this nicer look too

```c
  uint32_t x;
  uint16_t y = x(15, 0); // y = x[15:0]
  uint1_t b1 = x(3); // b1 = x[3];
  uint1_t b2 = x(3, 3); // b2 = x[3:3];
  float f;
  float_8_23_t femt;
  uint32_t f_as_u32 = f(31, 0); // f_as_u32 = f[31:0]
  uint32_t femt_as_u32 = femt(31, 0); // f_as_u32 = f[31:0] 
```

closed 2022-07-12 by JulianKemmerer

Quarky93, 2022-07-12:
This looks great! thanks

suarezvictor, 2022-07-12:
Pretty good! paves the way for a uture the templated version that requires compile time constants too