JulianKemmerer/PipelineC #109: Premature optimization

Datavenia, 2022-09-04:
Right now we have areas of code that are intended to be performant, but we don't have benchmarks/profiling in place to know where our hot spots are/whether something is a regression or performance increase.

I suggest for now we allow performance regression and strive for more idiomatic code over performant code. The mypy effort should alleviate performance issues in the long term (mypyc allows 4x speed ups over normal python) and once the code is cleaner in general we can resume optimization with more focus.

Thoughts?

JulianKemmerer, 2022-09-04:
I would say leave code as is if it seen done for performance. It should be one of the last things to change.

Focus your changes on easy code than doesn't look highly optimized + new features

JulianKemmerer, 2022-09-04:
Also having examples to point to would be good

closed 2024-11-16 by JulianKemmerer