JulianKemmerer/PipelineC #124: Examples missing `#pragma main`

rachitnigam, 2022-10-13:
The `matrix_mult_` examples are missing `#pragma MAIN main` from the C files which causes the tool to immediately error out. Should those exist there by default?

JulianKemmerer, 2022-10-14:
Oh gosh ew - those are some old examples and may have rotted away a little

Thanks for bringing to attention! Will be taking a look...

rachitnigam, 2022-10-14:
Maybe it would be useful to have some continuous testing to keep the examples up to date! For example, we use [runt](https://github.com/rachitnigam/runt) to do this very lightweight but useful form of [expectation testing](https://www.cs.cornell.edu/~asampson/blog/turnt.html). Maybe sometime like that would be useful? Happy to help figure out how to set it up if that'll be useful!

JulianKemmerer, 2022-10-14:
@rachitnigam some sort of CI does sound like a good idea - sadly it hasn't been prioritized as of yet

I highly encourage you to continue any discussion on this general issue over here https://github.com/JulianKemmerer/PipelineC/issues/104

JulianKemmerer, 2022-10-14:
Fixed missing `MAIN` pragmas in recent commit :+1:

closed 2022-10-14 by JulianKemmerer