JulianKemmerer/PipelineC #125: Generating output using `--verilog`

rachitnigam, 2022-10-13:
I tried to generate Verilog output on a fresh ubuntu machine and ran into some problems. This is to document the problems and solutions. This information should probably be added to the wiki page somewhere

## TLDR
You need to install latest versions of `ghdl`, `yosys`, and `ghdl-yosys-plugin` in order for the `--verilog` flag to work. Versions of tools installed using `apt-get` will not work.
- I build the tag `v2.0.0` from the GHDL github repository
- I built the latest tag for `yosys`
- I installed the `ghdl-yosys-plugin`

## Errors
I encountered the following errors which signaled a problem:
1. GHDL: Missing `float` IEEE library. This is because older versions of GHDL do not have it.
2. yosys: Failed to load GHDL shared library. This is because the `ghdl-yosys-plugin` is needed

JulianKemmerer, 2022-10-14:
Wow thanks for the great info - I will absolutely be summarizing those points into the wiki page to help folks avoid the same issues in future. Thanks again!

JulianKemmerer, 2022-10-14:
Thanks again - updated the wiki with this great info :+1: 
https://github.com/JulianKemmerer/PipelineC/wiki/Running-the-Tool

closed 2022-10-14 by JulianKemmerer