JulianKemmerer/PipelineC #126: Clean and standardize PipelineC repo.

bartokon, 2022-10-14:
Repo readability is really low. I suggest to move all .h files to somekind of headers folder and iee, io, axi folders to merge them. Do we really need one extra layer-directory for dvi/dvi_pmod.c? fixed/q0_23.h?

JulianKemmerer, 2022-10-14:
Making a `headers` directory sounds fine to me
And then move dirs+files like you said `headers/axi/axis.h` for ex.
`dvi/` dir was to match `vga/` which did end up having a few files in it
`fixed/` is supposed to be filled in with various fixed point type implementations (if they dont get autogenerated otherwise :-/)

JulianKemmerer, 2022-10-17:
I might prefer `include` as dir name over `headers`
ex. `include/axi/axis.h`

Datavenia, 2023-01-18:
Can we roll this out?

JulianKemmerer, 2023-01-18:
I think it needs some clarification still @Datavenia 

Is the goal to move directories with .h files into `include/<name>/<file>.h` ?

I am not in a rush to get to this

But if you put together a PR with the change I can try and test it after the big cleanup PR thats also pending probably