JulianKemmerer/PipelineC #129: Function, type, variable, etc names must not conflict with C and VHDL reserved words

JulianKemmerer, 2022-10-21:
For C this makes sense and folks know C keywords well...

But here are the VHDL keywords to watch out for 

```
abs
configuration
impure
null
rem
type
access
constant
in
of
report
unaffected
after
disconnect
inertial
on
return
units
alias
downto
inout
open
rol
until
all
else
is
or
ror
use
and
elsif
label
others
select
variable
architecture
end
library
out
severity
wait
array
entity
linkage
package
signal
when
assert
exit
literal
port
shared
while
attribute
file
loop
postponed
sla
with
begin
for
map
procedure
sll
xnor
block
function
mod
process
sra
xor
body
generate
nand
pure
srl

buffer
generic
new
range
subtype

bus
group
next
record
then

case
guarded
nor
register
to

component
if
not
reject
transport

```

Datavenia, 2023-01-18:
I think a workaround for this could simply be adding some sort of modifier to the names when lowering, like if ever user defined name had "pc_" on the front there would be no conflicts.

JulianKemmerer, 2023-01-18:
Yeah some kind of mangling seems required, right on 👍

JulianKemmerer, 2026-09-26:
In the Python front end, locals, struct fields and function names are renamed automatically, but @enum member names are emitted as-is. A member named RELEASE builds and simulates natively, then GHDL rejects the generated package.

closed 2026-09-26 by JulianKemmerer