JulianKemmerer/PipelineC #131: Vivado 2022.2 does not need fixed_pkg_2008 compiled

JulianKemmerer, 2022-10-22:
Thanks Bartus,
``` 
in vivado 2022.2 you don't need these lines in read_vhdl.tcl (It fails when they are used, it also fails in pipelineC get timing phase): 
'add_files -norecurse /tools/Xilinx/Vivado/2022.2/scripts/rt/data/fixed_pkg_2008.vhd
set_property library ieee_proposed [get_files /tools/Xilinx/Vivado/2022.2/scripts/rt/data/fixed_pkg_2008.vhd
```

Need way of checking what Vivado version is installed, etc...

bartokon, 2022-10-22:
I could also modify Vivado package_xo.tcl to read read_vhdl.txt and ignore first two lines. I don't know how this will change behavior for tools version other than 2022.2. 
Tomorrow, I will try to make some changes to vitis_import example, so hw-emu will run for alveo without issues. 
If we get the ap_ctrl_chain protocol working, I can try with sw_emu and hls blackbox integration. 
Also in 2022.2 it is possible for emulating zynq ultrascale boards on x86 - so maybe devboard simulation?

JulianKemmerer, 2022-10-26:
Done in https://github.com/JulianKemmerer/PipelineC/commit/6fb3ab4d391aeb50edbded267c708169829651f7

closed 2022-10-26 by JulianKemmerer