JulianKemmerer/PipelineC #145: Make cache of pipelined basic operations

JulianKemmerer, 2022-11-03:
ex. `pipeline_period_cache/vivado/xc7a35ticsg324/syn/BIN_OP_EQ_uint32_t_uint32_t_0.75.delay`

Largest stage determines the functional 'sliced at point' percentage.

Very similar to existing [path_delay_cache](https://github.com/JulianKemmerer/PipelineC/tree/master/path_delay_cache) except with an added 0.75 to name meaning like 'sliced at the 75% point'  ~= inverse 1/0.75 = 1.33 stage pipeline

I think it helps with faster estimates
https://github.com/JulianKemmerer/PipelineC/issues/46

And also can help for searching to get to a specific fmax goal 
https://github.com/JulianKemmerer/PipelineC/issues/48

Planned to work well modeling @suarezvictor is doing too

JulianKemmerer, 2022-11-10:
As discussed above there is a now a directory with cached minimum period delays observed for pipelined low level funcs like in path delay cache

It will need to be built up through experiment or simply over time as designs go through the tool...

https://github.com/JulianKemmerer/PipelineC/commit/ecbe94a726c9c27928cbe8b2bbe1224e9c534904

closed 2022-11-10 by JulianKemmerer

JulianKemmerer, 2022-11-10:
@suarezvictor can be free to experiment with this by running the tool locally - I will try to get generate some cached numbers myself too as I have time

suarezvictor, 2022-11-10:
good progress! it would be much better if you can just upload the data