JulianKemmerer/PipelineC #150: force pipeline stages in for loops

suarezvictor, 2023-01-06:
It seems that in a foor loop, amount of stages for each iteration ca be fractional, it would be good to force at least one for iteration
```
for (i=0; i<10; i+=1)
{
    force_pipeline(); //or #pragma
    other_code();
}
```

JulianKemmerer, 2023-01-06:
Sounding very familiar with Vitis HLS style stuff:

`#pragma HLS pipeline`
ex.
```c
   for (int i = 0; i < size; i++) {
       #pragma HLS PIPELINE II=1
        out = in + inc;
    }
```

[image]

JulianKemmerer, 2023-01-06:
Also reminded of https://github.com/JulianKemmerer/PipelineC/issues/97

Where by specifying a number of loops and pragma pipeline = "~each loop iteration is pipelined" - you are almost specifying the number of pipeline stages to use directly...