JulianKemmerer/PipelineC #151: celing rounding in coarse slicing

suarezvictor, 2023-01-06:
It seems it can never met timing by floor rounding, see example log (it should start at 11)

```
pixel_logic Path delay: 271.003 ns (3.690 MHz)
Function: pixel_logic Target MHz: 40.0
pixel_logic : sliced coarsely ~= 10 clocks latency...
```

JulianKemmerer, 2023-01-06:
271.003 ns  split into  25ns (40MHz) chunks == 10.84012  chunks
Rounded up to 11.

So there should be 11 stages = 10 clocks
(1 stage == comb logic == 0 clocks)

JulianKemmerer, 2023-01-06:
For now I dont think this is an issue

It doesnt seem like designs have been off by one when trying to meet timing - still needs iteration and ends up being more than 1 clock added anyway - so again dont think is a big issue or issue at all

closed 2023-01-06 by JulianKemmerer

suarezvictor, 2023-01-06:
The aim of this is to not to lose the first try since I never see it meeting the timings

JulianKemmerer, 2023-01-06:
I am saying the first try rarely ever meets timing

And it doesnt meet timing at +1 stages - it typically needs to almost double the number of stages (i.e. initial guess isnt that accurate)

suarezvictor, 2023-01-06:
Understood, thanks