JulianKemmerer/PipelineC #152: 1024x768 resolution

suarezvictor, 2023-01-06:
This timing set matches proves to work and matches LiteX's
```
#define PIXEL_CLK_MHZ 65.0

#define H_FP 24 //H front porch width (pixels)
#define H_PW 136 //H sync pulse width (pixels)
#define H_MAX 1344 //(2*(H_FP+H_PW)+FRAME_WIDTH) //H total period (pixels: active + blanking)
#define V_FP 3 //V front porch width (lines)
#define V_PW 6 //V sync pulse width (lines)
#define V_MAX 806 //(38+FRAME_HEIGHT) //V total period (lines: active + blanking)

#define H_POL 0
#define V_POL 0

```

suarezvictor, 2023-01-06:
Refences:
https://github.com/enjoy-digital/litex/blob/9115db5023266749c77b7b89ac5d8025ea8a5631/litex/soc/cores/video.py
http://tinyvga.com/vga-timing/1024x768@60Hz

JulianKemmerer, 2023-01-08:
Added in https://github.com/JulianKemmerer/PipelineC/commit/fe22dc6c9ba60766c553587d264dec5705745031

closed 2023-01-08 by JulianKemmerer