JulianKemmerer/PipelineC #166: Only 'cpp' for C preprocessing is supported now

JulianKemmerer, 2023-02-15:
Add support for other preprocessors, ex. `clang -E` perhaps?

Relates to https://github.com/JulianKemmerer/PipelineC/issues/56

A few places in code 'cpp' is hard coded...

JulianKemmerer, 2023-03-17:
Need to figure out what the clang equivalent of `cpp -MM -MG ` is

RossBencina, 2025-08-02:
Any reason not to use pcpp as the builtin fallback? https://pypi.org/project/pcpp/

I have used it on a personal project. Originally David Beazley developed it for SWIG I think, so it has some lineage.

JulianKemmerer, 2025-08-04:
I know of no reason not to, would be open to PRs 👍 

might touch on some other issues of how to handle pipelinec requirements
I think would just use it if its installed, otherwise old error message (as opposed to it being a required extra package?)