JulianKemmerer/PipelineC #180: Support full C syntax for invoking derived FSMs

JulianKemmerer, 2023-09-17:
Ex. consider a derived FSM function `uint32_t add1(uint32_t)` invoked three times sequentially:
```c
uint32_t add1(uint32_t x)
{
  uint32_t rv = x + 1;
  __clk();
  return rv;
}

uint32_t test1(uint32_t x)
{
  uint32_t rv = add1(add1(add1(x)));
  return rv;
}
```

The `add1(add1(add1(x` nesting syntax is not yet supported (requires logic to follow chain of ~almost recursive looking use of a function and allocate registers as needed).

Will get an error like:
`Exception: TODO unsupported control flow in func call argument:`  or similar...

Work around is to declare the intermediate variables yourself:
```c
a = add1(x)
b = add1(a);
c = add1(b);
//etc
```

RossBencina, 2025-08-02:
What do you envisage the fix would be? Do you just want to flatten the AST into a sequence of statements and temporary variables, something like:

```
uint32_t rv = a(b(c(x)));
```

transforms to:

```
uint32_t rv;
{
  T1 t1 = a(x);
  T2 t2 = b(t1);
  rv = c(t2);
}
```
This can be coded as an AST -> AST transformation pass.

The main complexity in my mind is you have to have access to enough type information to lookup types `T1` and `T2` from the signatures of `a` and `b`. Do you have that? I haven't looked at your code.

Or is the problem more that your current code can't recursively walk expression trees for some reason?

JulianKemmerer, 2025-08-04:
that transform makes sense to me 

the reason this issue/limit exists is that all the code for derived FSM style designs ( __clk(); func etc) inside C_TO_FSM.py is a relatively recent PipelineC code gen hack. (PipelineC derived FSM style -> 'hdl style' 'regular' PipelineC)

because of that many many initial assumptions were made to walk the ast in a completely different way than in C_TO_LOGIC.py,

I was open doing that proposed sort of rewrite as part of the PipelineC code gen already being done. But sounds reasonable to do it as a pycparser as transform too.

there is not really much if any infrastructure for ast level transforms. starting from me not knowing transforms and such. And now, knowing maybe pycparser is maybe not the best solution, I fret to put alot of time into manipulating pycparser C ast, not sure best practice for if ever wanted to support similar transforms for other front ends.

that said there is some very basic ast basic variable rename kind of manip done in C_TO_FSM.py probably done in bad ways ...