JulianKemmerer/PipelineC #191: Type not generated in c_structs_pkg.pkg.vhd

mayawarrier, 2024-02-17:
Hello, I'm a relative beginner at FPGAs, so please excuse me if the error/solution is obvious!

When a C struct becomes large enough, its VHDL type doesn't seem to be generated in c_structs_pkg. Here is the code that fails for me. When the bit width of `avsdr_in.writedata` is increased to 288 or above, the `avsdr_in` type fails to be defined in c_structs_pkg, causing a Quartus compilation error down the line complaining that avsdr_in is not defined.
```
// Input from SDRAM controller
typedef struct avmm_in
{
    uint1_t reset; 
    uint16_t readdata;
    uint1_t readdatavalid;
    uint1_t waitrequest;
} 
avmm_in;

typedef struct avsdr_in
{
    avmm_in av;
    uint1_t readstart;
    uint1_t writestart;
    uint32_t baseaddr;
    uint30_t nelems;
    uint256_t writedata;
}
avsdr_in;
```
Is the size limit intentional? Is there any way to work around this?

JulianKemmerer, 2024-02-17:
Hey there :wave: - feel free to stop by the pipelinec discord to for more general beginner chat :nerd_face: too if you like

Interesting you found it to be 288 as the limit - would have guesses >256 would be problem given bug.

The intentional current "limit" (purely to make for smaller vhdl files) is supposed to be 2048.

Just fixed bug where 256 was used instead - you should be good to go up to `u/int2048_t`. 
https://github.com/JulianKemmerer/PipelineC/commit/f1d41383ac6e1fe5a9fa642621ceddf0589ba459

Probably I recommend using arrays of bytes for things that arent literally 'a 2048b number to do math with etc', ex `uint8_t [256]`.

Hope things are working now - thanks.

closed 2024-02-17 by JulianKemmerer

mayawarrier, 2024-02-17:
@JulianKemmerer I had only tested with 288 and above (the actual type was selected using some macros) so I wasn't sure. Thanks a lot for the quick fix!