JulianKemmerer/PipelineC #285: Blink Example Fails

playduck, 2024-12-01:
Trying to build the `blink.c` example (or any other example I've tried) [as described in the wiki](https://github.com/JulianKemmerer/PipelineC/wiki) doesn't work:

```bash
% pipelinec examples/blink.c

██████╗ ██╗██████╗ ███████╗██╗     ██╗███╗   ██╗███████╗ ██████╗
██╔══██╗██║██╔══██╗██╔════╝██║     ██║████╗  ██║██╔════╝██╔════╝
██████╔╝██║██████╔╝█████╗  ██║     ██║██╔██╗ ██║█████╗  ██║
██╔═══╝ ██║██╔═══╝ ██╔══╝  ██║     ██║██║╚██╗██║██╔══╝  ██║
██║     ██║██║     ███████╗███████╗██║██║ ╚████║███████╗╚██████╗
╚═╝     ╚═╝╚═╝     ╚══════╝╚══════╝╚═╝╚═╝  ╚═══╝╚══════╝ ╚═════╝

Output directory: /.../PipelineC/pipelinec_output_blink.c_52928
================== Parsing C Code to Logical Hierarchy ================================
Parsing: /.../PipelineC/examples/blink.c
Preprocessing file...
/.../PipelineC/examples/blink.c:1:23: warning: extra tokens at end of #include directive [-Wextra-tokens]
    1 | #include "uintN_t.h"  // uintN_t types for any N
      |                       ^
      |                       //
1 warning generated.
Parsing C syntax...
pycparser says you messed up: /.../PipelineC/include/uintN_t.h:2:1: before: /
```

I'm assuming this project uses `Python3.9`, as it relies on `distutils`, however I don't think the documentation mentions this anywhere. I had to alias my `python3` command.
Running `pipelinec --help` works as expected and exits without error.
Perhaps I'm doing something wrong/my environment is wrong.

Versions:
- PipelineC commit [`28786db`](https://github.com/JulianKemmerer/PipelineC/commit/28786db4bd0024ec07ada00f261aec4d6456cfb5) (current head)
- `python3.9 --version`: Python 3.9.20
- MacOS 15.1.1 (Apple M4)
- `cpp --version`: Apple clang version 16.0.0 (clang-1600.0.26.4) (Target: arm64-apple-darwin24.1.0)

Additional interesting outputs from examples:
```bash
% ./src/pipelinec ./examples/adder_tree.c

██████╗ ██╗██████╗ ███████╗██╗     ██╗███╗   ██╗███████╗ ██████╗
██╔══██╗██║██╔══██╗██╔════╝██║     ██║████╗  ██║██╔════╝██╔════╝
██████╔╝██║██████╔╝█████╗  ██║     ██║██╔██╗ ██║█████╗  ██║
██╔═══╝ ██║██╔═══╝ ██╔══╝  ██║     ██║██║╚██╗██║██╔══╝  ██║
██║     ██║██║     ███████╗███████╗██║██║ ╚████║███████╗╚██████╗
╚═╝     ╚═╝╚═╝     ╚══════╝╚══════╝╚═╝╚═╝  ╚═══╝╚══════╝ ╚═════╝

Output directory: /.../PipelineC/pipelinec_output_adder_tree.c_56776
================== Parsing C Code to Logical Hierarchy ================================
Parsing: /.../PipelineC/examples/adder_tree.c
Preprocessing file...
Parsing C syntax...
pycparser says you messed up: /.../PipelineC/examples/adder_tree.c:2:1: before: /
```
```bash
% ./src/pipelinec ./examples/fir.c

██████╗ ██╗██████╗ ███████╗██╗     ██╗███╗   ██╗███████╗ ██████╗
██╔══██╗██║██╔══██╗██╔════╝██║     ██║████╗  ██║██╔════╝██╔════╝
██████╔╝██║██████╔╝█████╗  ██║     ██║██╔██╗ ██║█████╗  ██║
██╔═══╝ ██║██╔═══╝ ██╔══╝  ██║     ██║██║╚██╗██║██╔══╝  ██║
██║     ██║██║     ███████╗███████╗██║██║ ╚████║███████╗╚██████╗
╚═╝     ╚═╝╚═╝     ╚══════╝╚══════╝╚═╝╚═╝  ╚═══╝╚══════╝ ╚═════╝

Output directory: /.../PipelineC/pipelinec_output_fir.c_55965
================== Parsing C Code to Logical Hierarchy ================================
Parsing: /.../PipelineC/examples/fir.c
Traceback (most recent call last):
  File "/.../PipelineC/./src/pipelinec", line 220, in <module>
    parser_state = C_TO_LOGIC.PARSE_FILE(c_file)
  File "/.../PipelineC/src/C_TO_LOGIC.py", line 9986, in PARSE_FILE
[... 92 lines cut ...]
   94 | DECL_INPUT(uint4_t, s_axis_tkeep)
      | ^
/.../PipelineC/include/compiler.h:97:14: note: expanded from macro 'DECL_INPUT'
   97 | type_t name; \
      |              ^
/.../PipelineC/include/compiler.h:66:27: note: expanded from macro '\
PRAGMA_MESSAGE'
   66 | #define PRAGMA_MESSAGE(x) PRAGMA_MESSAGE_(x)
      |                           ^

[removed additional output for brevity, more of the same pragma errors as above]

fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.

handshake_io.o: \
  /.../PipelineC/examples/handshake_io.c \
  /.../PipelineC/include/arrays.h \
  /.../PipelineC/include/handshake/handshake.h \
  /.../PipelineC/include/compiler.h \
  /.../PipelineC/include/bool.h \
  /.../PipelineC/include/uintN_t.h \
  /.../PipelineC/include/stream/stream.h

```

JulianKemmerer, 2024-12-01:
Thanks for pointing this out.

Has been some reorganization lately, hoping it's something simple but will
see if it's something funny related to python versions 🤞

JulianKemmerer, 2024-12-01:
Ah OK - I missed the part about this being on Mac.

I am not able to recreate this on my Linux system.

PipelineC has only ever been tested on Linux, so I am just starting to speculate about what could be going on for you.  (and any python3 version should work fine out of the box, why not specified specific version for non-developers).

```
/.../PipelineC/examples/blink.c:1:23: warning: extra tokens at end of #include directive [-Wextra-tokens]
    1 | #include "uintN_t.h"  // uintN_t types for any N
      |                       ^
      |                       //
1 warning generated.
Parsing C syntax...
pycparser says you messed up: /.../PipelineC/include/uintN_t.h:2:1: before: /
```

Seems like something is wrong dealing with include statements during preprocessing. Is `cpp` on Mac still the C preprocessor or something else?

At one point there was an issue of some `cpp` versions not having the same cmd line args: https://github.com/JulianKemmerer/PipelineC/issues/166

I am going to make a general issue for 'pipelinec doesnt work on mac' and tag you in it, continue investigating there.

In the short term using PipelineC on a Linux machine, virtual even, is the recommended fix.

playduck, 2024-12-01:
Right, I just tested it again in a Ubuntu 20.04 VM (via UTM, virtualized so still arm-based) and the `blink` example works as expected.

This is likely an issue with Apple's `clang` not being 1:1 compatible with the `cpp` compiler.
Definitely a viable workaround, still a bit weird. Would be interesting to know why `clang` seems to have problems with its preprocessor.

closed 2024-12-01 by playduck