JulianKemmerer/PipelineC #291: Install

csantosb, 2025-03-16:
Hi,

I just did a git clone, followed by a `pip install .`. I get the following error:

    poetry.core.masonry.utils.module.ModuleOrPackageNotFoundError: No file/folder found for package pipelinec

Am I doing something wrong ? 

Thanks,

JulianKemmerer, 2025-03-17:
Hi there

First off, to use pipelinec no pip installs should be required, just clone and run like [wiki](https://github.com/JulianKemmerer/PipelineC/wiki) says 
```
# Install 'pipelinec' executable on Linux
git clone https://github.com/JulianKemmerer/PipelineC.git
cd PipelineC/
export PATH=$PATH:$(pwd)/src
# Run blink example
pipelinec ./examples/blink.c
```

If you want to contribute to the pipelinec project, yes there is some poetry based setup and I think is talked about [here](https://github.com/JulianKemmerer/PipelineC/blob/master/CONTRIBUTING.md#pre-commit) 

let me know if I can help any more 👋

csantosb, 2025-03-17:
Well, I aim at packaging pipelinec under [Guix](https://guix.gnu.org/). I'm wondering what the best strategy would be, in this case.

JulianKemmerer, 2025-03-17:
I also wonder the best strategy. I am not expert on how to package python projects. Luckily the project so far doesn't really use any non standard python packages so have been getting away with 'just clone and run' / everything is already installed on most Linux distros.

I am hearing of Guix now

have heard of conda things?
pip based installs?
docker images?
nix?

they all seem alittle complicated and non are familiar to me. So I am very open to your suggestions. There is a #dev channel on the pipelinec discord too if that works for you https://discord.gg/Aupm3DDrK2 some folks have had similar thoughts in the past.

JulianKemmerer, 2025-03-17:
I am going to move this from an issue to a discussion on how to package pipelinec